JuliaRobotics/IncrementalInference.jl #1811: test issue, no method mean!

dehann, 2024-01-21:
```julia
: Error During Test at /home/runner/work/IncrementalInference.jl/IncrementalInference.jl/test/testpartialconstraint.jl:321
  Test threw exception
  Expression: 0 < var(getManifold(X2), getPoints(X2))
  MethodError: no method matching mean!(::TranslationGroup{ManifoldsBase.TypeParameter{Tuple{2}}, ℝ}, ::MVector{2, Float64}, ::Vector{SVector{2, Float64}}, ::StatsBase.UnitWeights{Float64}, ::EfficientEstimator)
  
  Closest candidates are:
    mean!(::AbstractManifold, ::Any, ::AbstractVector, ::AbstractVector, ::GeodesicInterpolation; shuffle_rng, retraction, inverse_retraction, kwargs...)
     @ Manifolds ~/.julia/packages/Manifolds/A0dYY/src/statistics.jl:256
    mean!(::AbstractManifold, ::Any, ::AbstractVector, ::AbstractVector, ::GradientDescentEstimation; p0, stop_iter, retraction, inverse_retraction, kwargs...)
     @ Manifolds ~/.julia/packages/Manifolds/A0dYY/src/statistics.jl:189
    mean!(::AbstractManifold, ::Any, ::AbstractVector, ::AbstractVector, ::ExtrinsicEstimation; extrinsic_method, kwargs...)
     @ Manifolds ~/.julia/packages/Manifolds/A0dYY/src/statistics.jl:404
    ...
  
  Stacktrace:
    [1] mean(M::TranslationGroup{ManifoldsBase.TypeParameter{Tuple{2}}, ℝ}, x::Vector{SVector{2, Float64}}, w::StatsBase.UnitWeights{Float64}, method::EfficientEstimator; kwargs::@Kwargs{})
      @ Manifolds ~/.julia/packages/Manifolds/A0dYY/src/statistics.jl:161
    [2] mean(M::TranslationGroup{ManifoldsBase.TypeParameter{Tuple{2}}, ℝ}, x::Vector{SVector{2, Float64}}, w::StatsBase.UnitWeights{Float64}, method::EfficientEstimator)
      @ Manifolds ~/.julia/packages/Manifolds/A0dYY/src/statistics.jl:153
    [3] mean_and_var(M::TranslationGroup{ManifoldsBase.TypeParameter{Tuple{2}}, ℝ}, x::Vector{SVector{2, Float64}}, w::StatsBase.UnitWeights{Float64}, method::EfficientEstimator; corrected::Bool, kwargs::@Kwargs{})
      @ Manifolds ~/.julia/packages/Manifolds/A0dYY/src/statistics.jl:842
    [4] mean_and_var(M::TranslationGroup{ManifoldsBase.TypeParameter{Tuple{2}}, ℝ}, x::Vector{SVector{2, Float64}}, method::EfficientEstimator; corrected::Bool, kwargs::@Kwargs{})
      @ Manifolds ~/.julia/packages/Manifolds/A0dYY/src/statistics.jl:859
    [5] mean_and_var(M::TranslationGroup{ManifoldsBase.TypeParameter{Tuple{2}}, ℝ}, x::Vector{SVector{2, Float64}}, method::EfficientEstimator)
      @ Manifolds ~/.julia/packages/Manifolds/A0dYY/src/statistics.jl:846
    [6] #var#128
      @ Manifolds ~/.julia/packages/Manifolds/A0dYY/src/statistics.jl:795 [inlined]
    [7] var(M::TranslationGroup{ManifoldsBase.TypeParameter{Tuple{2}}, ℝ}, x::Vector{SVector{2, Float64}})
      @ Manifolds ~/.julia/packages/Manifolds/A0dYY/src/statistics.jl:794
    [8] macro expansion
      @ /opt/hostedtoolcache/julia/1.10.0/x64/share/julia/stdlib/v1.10/Test/src/Test.jl:669 [inlined]
    [9] macro expansion
      @ ~/work/IncrementalInference.jl/IncrementalInference.jl/test/testpartialconstraint.jl:321 [inlined]
   [10] macro expansion
      @ /opt/hostedtoolcache/julia/1.10.0/x64/share/julia/stdlib/v1.10/Test/src/Test.jl:1577 [inlined]
   [11] top-level scope
      @ ~/work/IncrementalInference.jl/IncrementalInference.jl/test/testpartialconstraint.jl:257
Test Summary:                           | Pass  Error  Broken  Total  Time
test belief prediction with partials... |   10      1       4     15  2.9s
ERROR: LoadError: Some tests did not pass: 10 passed, 0 failed, 1 errored, 4 broken.
in expression starting at /home/runner/work/IncrementalInference.jl/IncrementalInference.jl/test/testpartialconstraint.jl:253
in expression starting at /home/runner/work/IncrementalInference.jl/IncrementalInference.jl/test/runtests.jl:13
ERROR: LoadError: Package IncrementalInference errored during testing
Stacktrace:
 [1] pkgerror(msg::String)
   @ Pkg.Types /opt/hostedtoolcache/julia/1.10.0/x64/share/julia/stdlib/v1.10/Pkg/src/Types.jl:70
 [2] test(ctx::Pkg.Types.Context, pkgs::Vector{Pkg.Types.PackageSpec}; coverage::Bool, julia_args::Cmd, test_args::Cmd, test_fn::Nothing, force_latest_compatible_version::Bool, allow_earlier_backwards_compatible_versions::Bool, allow_reresolve::Bool)
   @ Pkg.Operations /opt/hostedtoolcache/julia/1.10.0/x64/share/julia/stdlib/v1.10/Pkg/src/Operations.jl:2018
 [3] test
   @ /opt/hostedtoolcache/julia/1.10.0/x64/share/julia/stdlib/v1.10/Pkg/src/Operations.jl:1899 [inlined]
 [4] test(ctx::Pkg.Types.Context, pkgs::Vector{Pkg.Types.PackageSpec}; coverage::Bool, test_fn::Nothing, julia_args::Vector{String}, test_args::Cmd, force_latest_compatible_version::Bool, allow_earlier_backwards_compatible_versions::Bool, allow_reresolve::Bool, kwargs::@Kwargs{io::IOContext{Base.PipeEndpoint}})
   @ Pkg.API /opt/hostedtoolcache/julia/1.10.0/x64/share/julia/stdlib/v1.10/Pkg/src/API.jl:444
 [5] test(pkgs::Vector{Pkg.Types.PackageSpec}; io::IOContext{Base.PipeEndpoint}, kwargs::@Kwargs{coverage::Bool, julia_args::Vector{String}, force_latest_compatible_version::Bool})
   @ Pkg.API /opt/hostedtoolcache/julia/1.10.0/x64/share/julia/stdlib/v1.10/Pkg/src/API.jl:159
 [6] test(; name::Nothing, uuid::Nothing, version::Nothing, url::Nothing, rev::Nothing, path::Nothing, mode::Pkg.Types.PackageMode, subdir::Nothing, kwargs::@Kwargs{coverage::Bool, julia_args::Vector{String}, force_latest_compatible_version::Bool})
   @ Pkg.API /opt/hostedtoolcache/julia/1.10.0/x64/share/julia/stdlib/v1.10/Pkg/src/API.jl:174
 [7] top-level scope
   @ ~/work/_actions/julia-actions/julia-runtest/v1/test_harness.jl:15
 [8] include(fname::String)
   @ Base.MainInclude ./client.jl:489
 [9] top-level scope
   @ none:1
in expression starting at /home/runner/work/_actions/julia-actions/julia-runtest/v1/test_harness.jl:7
```

dehann, 2024-01-21:
Potential Manifolds.jl internal new bug.  It looks like while computing the `mean`, a method is passed implicitly for `default_approximation_method(M, mean, eltype(x))` which returns `EfficientEstimator`, but that dispatch no longer seems available.  Error suggests to instead use `Geodesic / Gradient / ExtrinsicEstimation`.

Hi @mateuszbaran, do you perhaps have any suggestions -- I think this is a new integration bug.

mateuszbaran, 2024-01-21:
Hi! I will look into it in a few hours. We probably lack some method in Manifolds.jl.

mateuszbaran, 2024-01-21:
It looks like the best solution is to just define `var` for `TranslationGroup` which cuts out a lot of compilation but isn't totally generic. I've made a pull request.

closed 2024-01-22 by dehann